M17-Project/Module_17 #7: Digital potentiometer out of stock everywhere

ManoDaSilva, 2022-11-20:
Hi! 

Amazing work you've been doing on developing this modem to get as many people as possible going with M17! I'm preparing an order with JLCPCB, with assembled components, etc.  I'm now looking for the digital potentiometers, however it's out of stock everywhere. Is there a suitable replacement I can look for? Or is there a way to replace it with a physical potentiometer to manually adjust the filters? How does openRTX interact with those? 

Thanks for the help!

silseva, 2022-11-20:
Hi!
Digital potentiometers are used to regulate the level of the audio signal to and from the radio. You can replace those with physical ones without problems.

73,
Silvano IU2KWO

sp5wwp, 2022-11-29:
My proposal is to use 50k ones for both `U8` and `U9`. 50k ones are available at Mouser. And change the `R42` to 10k.

sp5wwp, 2023-05-16:
Done.

closed 2023-05-16 by sp5wwp