M17-Project/Module_17 #15: License update

mdiepart, 2023-11-19:
We have been discussing a license change for this project. Mainly, we want to change from the TAPR Open Source Hardware License to the CC-BY-NC-SA license (version 4).

For now, the following contributors expressed their agreement with this change:
@sp5wwp 
@kc1awv 
@mathisschmieder 
@tarxvftech 

We also have identified the following contributors but haven't asked your opinion yet:
@AC4AG (under the usernames AC4AG and popracer) that contributed to the 3D printed cases for revs 0.1d and 0.1e.
@abfparts that also contributed to the 3D printed cases
@n1zzo that contributed to the drawio diagram.

Can I please ask everybody, to confirm if they agree about this license change? And more especially, could those who we haven't asked about this yet please tell us if they agree or not?

mdiepart, 2023-11-19:
To be more accurate, we want to change the license of the repository for the revision 1 mainly. But everything in the repository will be under that license once the change is done. The previous revisions will still be under TAPR OSHL.

tarxvftech, 2023-11-19:
I agree. I don't think I actually have copyrightable contributions but I appreciate the attention to detail in this license change.

mdiepart, 2023-12-06:
It have finally been decided to let go of the NC clause mostly because we believe the manufacturers that want to manufacture the board will do it irrespective of the license we choose. NC clause is also refused by OSHW for their certification.
It was decided with @sp5wwp, @kc1awv and @mathisschmieder to opt for a CERN-OHL-V2-W (https://ohwr.org/cern_ohl_w_v2.txt).

Can I please ask you guys (@AC4AG, @abfparts and @n1zzo) to say here if you agree or disagree with this proposed change?
Please not that we do not ask you to bother you. You have contributed to this repository so we NEED your agreement to re-license your contributions.

If we have not received any answers by 10/12/2023 those changes will be moved to another repository.

mdiepart, 2023-12-08:
Amazing, thank you very much!

n1zzo, 2023-12-08:
I agree about the license change!

tarxvftech, 2023-12-09:
I agree. Furthermore if there _is_ anything in here that's actually mine, I assign copyright and ownership of everything of mine in this repo up to now to Steve KC1AWV.

mdiepart, 2023-12-09:
Thank you all for your replies, it is much appreciated!

closed 2023-12-09 by mdiepart