M17-Project/Module_17 #16: Add some more clearance around mounting holes

dd5xl, 2024-02-03:
Clearance around 3 of 4 mounting holes is not sufficient to add washer or to use a wrench to tighten the nuts.

[image]

In case of a PCB redesign please consider to add some more clearance.

sp5wwp, 2024-06-09:
1.0 is a complete redesign, so it should be OK then :)

mdiepart, 2024-07-28:
Revision 1.0 has at least 1.8mm of clearance between the edge of the hole and the closest component. This clearance can be increased if opting for digital potentiometers instead of the multiturn potentiometers. Revision1.0 is also designed with a specific enclosure in mind.

closed 2024-07-28 by mdiepart