M17-Project/Module_17 #17: Pinout of DB-9 can be wrong

pe1msz, 2024-02-08:
I noticed that a board I got in, has a female connector mounted, which causes some pin-numbers to be wrong.

I use a GM950 Motorola, and that seems to work great sofar (it has a data transmit and a flat audio out).

I am just wondering what the best RX-Post value is on the M17-module (allignable with a multiturn pot)

sp5wwp, 2024-06-09:
Which pins are wrong?

For the calibration procedure, see [this](https://github.com/OpenRTX/OpenRTX/issues/184).

pe1msz, 2024-06-11:
Pinnumbering on the DB9 was not the same as on the connector.

 

I will get more details for you.

mdiepart, 2024-07-28:
Is this issue solved?

kc1awv, 2024-08-05:
> I noticed that a board I got in, has a female connector mounted, which causes some pin-numbers to be wrong.

The pin numbers are always going to be the same, regardless of the connector's gender. If you're referring to the wiring diagrams provided in the repo, remember that the diagrams describe the _cable_, and not the radio it's plugging into.

pe1msz, 2024-08-06:
You are right, the pins are correct! I looked at the pinout of the connector which is in the doc-folder, and that shows the front of a male connector. The board has a female connector, so i believe it would be useful to show the female connector. In my case I had the idea that the board had the wrong connector, thus having a wrong pinout.

pe1msz, 2024-08-12:
I agree, but if you use the numbers on the picture, they are reversed as it is not the back but the front of the connector (male)

sp5wwp, 2024-08-31:
It says "male" in the image. Closing.

closed 2024-08-31 by sp5wwp