M17-Project/Module_17 #18: Short circuit VDD to Ground

DL6YA, 2024-05-25:
I uploaded the files for version r0.1e to JLCPCB the way it was described. All 10 boards have a short circuit fromm VDD to Ground. JLCPCB told me that the file for the NPTH holes should not be on the copper layer? Did someone had similar problems? Now I am afraid burning another 300$ for useless boards. 
[image]

sp5wwp, 2024-06-09:
We are very sorry to hear this. We have ordered multiple boards without facing any issues. Your image did not upload, can you try again? We need more detailed info.

DL6YA, 2024-07-11:
Hey, it was just a production mistae. Got a new production and they work so far. Still didnt manage to make a qso, but our group is working on the cbale for there radios. hopfully we will report good news soon. are there any information about the adjustment for Rx an Tx?

mdiepart, 2024-07-28:
Given that this was a production mishap and that the current revision is now revision 1.0, I'm gonna close this issue. 

Feel free to re-open if this happens on the new revision too :)

closed 2024-07-28 by mdiepart