M17-Project/Module_17 #21: TX adjustment could be clearer

chichimus, 2024-09-02:
Maybe I was having an off day but the directions on adjusting the TX could be clarified. I picked up a board and was trying it out with another ham.  I took the "_While pressing the PTT, have a look at the symbols graph of the m17_decoder module and adjust the baseband TX level until the blue dots line-up with the gray lines in the background_" to mean all the blue dots in a single blue line mid screen and lined up with the grey lines. This was fairly quickly achieved with the direction I first turned the pot and thought I was good to go. Was getting nowhere until my friend sent me a screenshot of his and now I have it adjusted properly.

sp5wwp, 2024-10-18:
Tell us how to reword it to make it clearer.

sp5wwp, 2024-10-21:
Please open a PR with proposed changes.

closed 2024-10-21 by sp5wwp