M17-Project/Module_17 #22: Module17_Rev_1.0.zip appears to be broken

kandev, 2024-10-21:
I have just tried to follow the PCB ordering guide, but the gerber files zip archive is broken and can't be opened.

kandev, 2024-10-21:
Actually all the zip files are broken!? Are they zip files at all?

mdiepart, 2024-10-21:
Hello,

I just tried opening those files and it works without any issue. I know several people did so too without any issue either.

We will need more details on the problems you are having but it probably does not come from the gerber archives.

sp5wwp, 2024-10-21:
Opens with 7z with no problems (under Windows 10).

kandev, 2024-10-21:
Tried on both Windows (integrated zip, winrar and 7zip) and in Linux with unzip - same result always:

[image]

mdiepart, 2024-10-21:
Are you sure you did not download the html page as an archive?

sp5wwp, 2024-10-21:
It is an HTML page...

mdiepart, 2024-10-21:
The link you need to use to download the archive via wget is this one: https://github.com/M17-Project/Module_17/raw/refs/heads/main/hardware/mainboard/production/Module17_Rev_1.0.zip

Closing as it is a non-issue.

kandev, 2024-10-21:
Yup, that was it. The direct links to the files in the manual are not so direct :)
Thank you!